Koromix/tytools #98: TyUploader Doesn't detect Teensy MM in Serial+MTP mode

david-res, 2022-09-30:
I have a Teensy MicroMod set up as a Serial+MTP device, when plugged into USB, it does not appear in the device list.
If I click the boot button it appears as HalfKay
[image]

Is there a way to fix this?

Defragster, 2022-09-30:
There was a fix to ID the new VID_PID this Serial+MTP creates, but it hasn't been formally released ?

david-res, 2022-10-01:
@Defragster thanks for confirming!
What OS are you running on?
If you have a macos build I’d love to try it out.
@Koromix of you could create a new release with the fix that would be highly appreciated

Defragster, 2022-10-01:
Sorry - I'm on Windows - so no help there.

Koromix, 2022-10-01:
Yeah this is overdue, I don't have a Mac anymore but I'll try to make a release anyway 😁

Koromix, 2022-10-01:
Although I do have a running Mac on QEMU for another project, so that should do :) I'll make a release today then!

david-res, 2022-10-01:
@Koromix thanks! Very appreciated!

Koromix, 2022-10-01:
Not home right now, so it'll have to wait until tomorrow :)

Koromix, 2022-10-02:
I've made a release: https://github.com/Koromix/tytools/releases/tag/v0.9.8

closed 2022-10-02 by Koromix

david-res, 2022-10-03:
All working! Thanks!!