Koromix/tytools #100: Clipboard copy and Paste is not returning the full selection (0.9.8)

KurtE, 2022-10-03:
This morning I updated to the new release. 

As I mentioned on the Teensy Thread:
https://forum.pjrc.com/threads/27825-Teensy-Qt?p=313682&viewfull=1#post313682

I was debugging some code this morning.  With lots of stuff printed.
I selected all of the text, did a copy, went into sublimetext and did a paste and it only transferred the first about 176 lines.  Which my rough guess was about 1/3rd of what was selected.

I did not notice these issues with the previous release, including my simple extension for the new Product ID.

Thanks
Kurt

Koromix, 2022-11-25:
Not sure, did it happen again?
In theory nothing has changed with the live serial view. But it could be a Qt bug.

KurtE, 2022-11-26:
I have been getting some strange results with copy and paste.

It is working mostly better with the version I built with the sources that I synced up. 

However, at times I am still getting some strange results, 

Like I do a copy from it, and try to do a paste into Google Chat, and nothing shows up.  I then have a window open in sublibme text and often times it will allow me to paste.  I then reselect the text and paste it into goggle chat.  But other times it works fine.

Maybe it is something with Windows 11? 

Side note: in the last couple of days, I have run into some instances where, I program the Teensy using IDE2 with serial turned disabled, wait for that to complete, and then enable serial.   Then later I press the reset button in the TyTerminal window, and the Teensy app, tries to erase the flash and hangs.  If I then kill the Teensy app, the reset then works fine again.
But probably not related.

Defragster, 2023-03-01:
Win 11 here and T_4.1's running - copy text - paste one into Notepad - then all three into Sublimetext to count lines:

40,421 lines okay
101,243 lines okay
200,000 as set and all okay

As for RESET causing UPLOAD:
That is a RACE condition. When Teensy.exe sees the reset it may try to program.
> done twice it reset T_4.1 with just a twitch of icon change on Teensy.exe
> Done another time Teensy entered program mode and then stuck for timeout as the T_4.1 restarted and there was nothing to program

KurtE, 2023-03-19:
@Koromix @Defragster - I am wondering if maybe there is an issue with

KurtE, 2023-03-21:
Just a minor updated comment.  Right now not able to do a copy out of the current captured data.  It will will allow me to copy  some portions, but not all.  
[image]

You hopefully can see the area that I selected and neither ctrl+c nor context menu copy, allowed me to paste the information here or into sublime-text

Defragster, 2023-03-21:
@KurtE - isn't something like this showing in IDE 2 SerMon as well?

KurtE, 2023-03-23:
@Defragster - IDE 2s serial monitor has it's own problems....  Lots!  they did fix the problem where the autoscroll to the end did not work properly... (nightly builds)

@Koromix,  maybe some hints on at least the current manifestations of the issue I am seeing.   And again I don't know what things are actually important to reproduce.  But in this case two processors running and connected up.  The one that is currently screwing up is generic at 115200...  

If I start up TyCommander and for example reboot the one board which generates a reasonable amount of debug data, which is sent out over Serial1 and using the second board which for this case does Serial to USB... 

On the first run on just started TyCommander.
I can start copy/paste from start of buffer, but first line is garbage:
```
��AQ=BK2�չ��0x57f4 22
16 03 55 00 53 00 42 00 20 00 53 00 65 00 72 00  : ..U.S.B. .S.e.r.
69 00 61 00 6C 00                                : i.a.l.
$I 6a 60
	EP0 len:0 typ:80 cod:6 P:0x20000146
GET_DESCRIPTOR IN/EP0
$I 4a 40
$I 4a f0
	SUP: a1 21 7 0 0
$I 6a 60
	EP0 len:0 typ:a1 cod:21 P:0x20000563
$I 4a 40
	NSR: a1 21
```

So now I clear serial command (context menu), and hit the reset again:
The first line now does not look garbled.
[image]
But if I try to do a copy/paste starting at the first line:
It fails to cpy/paste anything.

But if I start at second line:
```
exit: USBFS_RCC_Init
enter: USBFS_Device_Init(1)
enter: USBFS_Device_Endp_Init
exit: USBFS_Device_Endp_Init
	0x15a8, 00 02 00
	0x15e8, 00 02 00
	0x1cc, 00 00 00
	0x15f8, 00 02 00
exit: USBFS_Device_Init
```

Again this is not the only way I have seen it,  I have had it for example where I can not copy/paste from a sub portion of the data starting later.  

Again not sure what is important here?  Multiple Serial objects?  Not being a Teensy?  ...

KurtE, 2023-05-05:
@Koromix @Defragster - I had it hit me again today. And I am wondering if maybe something is somewhat parsing the data and some key byte sequences might be the issue I am seeing:
[image]

But I only received 21 lines (actually the 21st was truncated:
```
$$r Small-Drake.jpg

CMD:3 Filename:Small-Drake.jpg
Downloading Small-Drake.jpg
lsrhid: 3 0x20057ee0 15 250
download failed
onReceiveDownload(ffab0200 0x2000461c 64) - Status: 0 54453
onReceiveDownload(ffab0200 0x200045dc 64) - Data cb:60 bufsize:60

#################### Enter Download main Loop ##########################

>>>>>> Enter data loop (60 0 0) <<<<<<<<
U-S(466 466):RLOOP CMD:3 15
Sending Small-Drake.jpg
RSHID CMD:7 0x20067f70 8 1000
RMT: 20003804 - 07 00 08 00 00 00 00 00  B5 D4 00 00 00 00 00 00  : ........ ........
20003814 - 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  : ........ ........
20003824 - 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  : ........ ........
20003834 - 00 00 00 00 00 00 00 00  00 00 00 00 00 00 00 00  : ........ ........
RSHID CMD:0 0x20003f04 60 5000
RMT: 20003804 - 00 00 3C 00 FF D8U-S(64 64):
```
So wondering if it looks for things like <E  or maybe  :<E 
for something?

Defragster, 2023-05-05:
Saw it the other day - thought I left a comment somewhere?

A print of a 0x0 - Zero value - will terminate copy select.

closed 2023-10-19 by KurtE