Koromix/tytools #115: TyUploader can’t open the Open panel on macOS

Peterbing, 2025-05-02:
Sorry if a dupe or if I missed this in a search. I am trying to update an m8 Tracker v1 to the latest firmware (6.0.0 as of this writing). I’ve updated this device a bunch of other times on this computer.

Today though, I can’t get TyUploader to open the Open panel. m8 appears in the drop-down menu, but when I press the “Upload” button, nothing happens, so I can’t select the firmware file to proceed. I’ve rebooted once and installed the latest tytools in case that was the issue.

I do Apple development for work so I poked around a bit. This message is logged to the console:
```
Unable to display open panel (unrecognized error: Error Domain=com.apple.ViewBridge Code=14 "(null)" UserInfo={com.apple.ViewBridge.error.hint=-[NSViewServiceMarshal bootstrap:withReply:] caught NSInvalidArgumentException '*** -[NSUserDefaults setObject:forKey:inDomain:]: domain name cannot be nil or empty' with backtrace (
    "0   CoreFoundation                      __exceptionPreprocess + 176",
    "1   libobjc.A.dylib                     objc_exception_throw + 88",
    "2   Foundation                          -[NSUserDefaults(NSUserDefaults) setObject:forKey:inDomain:] + 420",
    "3   AppKit                              +[NSOSPDefaults _readFileListModeForIdentifier:inSaveMode:inDirectory:lastListForMediaBrowserMode:] + 524",
    "4   AppKit                              -[NSRemoteSavePanel(NSSavePanelLayout) _makeFinderKitView] + 416",
    "5   AppKit                              -[NSRemoteSavePanel(NSSavePanelLayout) _setupFileBrowserView] + 68",
    "6   AppKit                              -[NSRemoteSavePanel(NSSavePanelLayout) _initContentView] + 1544",
    "7   AppKit                              -[NSRemoteSavePanel initWithContentRect:styleMask:backing:defer:] + 488",
    "8   AppKit                              -[NSRemoteSavePanel initWithOptions:] + 476",
    "9   com.apple.appkit.xpc.openAndSavePan main + 708",
    "10  com.apple.appkit.xpc.openAndSavePan com.apple.appkit.xpc.openAndSavePan + 6440",
    "11  AppKit                              -[NSViewController _loadViewIfRequired] + 72",
    "12  AppKit                              __24-[NSViewController view]_block_invoke + 28",
    "13  AppKit                              NSPerformVisuallyAtomicChange + 108",
    "14  AppKit                              -[NSViewController view] + 160",
    "15  ViewBridge                          -[NSViewServiceMarshal _bootstrap:replyData:completion:] + 1308",
    "16  ViewBridge                          -[NSViewServiceMarshal bootstrap:withReply:] + 272",
    "17  CoreFoundation                      __invoking___ + 148",
    "18  CoreFoundation                      -[NSInvocation invoke] + 424",
    "19  CoreFoundation                      -[NSInvocation invokeWithTarget:] + 64",
    "20  ViewBridge                          -[NSVB_ViewServiceImplicitAnimationDecodingProxy forwardInvocation:] + 192",
    "21  CoreFoundation                      ___forwarding___ + 956",
    "22  CoreFoundation                      _CF_forwarding_prep_0 + 96",
    "23  CoreFoundation                      __invoking___ + 148",
    "24  CoreFoundation                      -[NSInvocation invoke] + 424",
    "25  CoreFoundation                      -[NSInvocation invokeWithTarget:] + 64",
    "26  ViewBridge                          -[NSVB_QueueingProxy forwardInvocation:] + 308",
    "27  CoreFoundation                      ___forwarding___ + 956",
    "28  CoreFoundation                      _CF_forwarding_prep_0 + 96",
    "29  CoreFoundation                      __invoking___ + 148",
    "30  CoreFoundation                      -[NSInvocation invoke] + 424",
    "31  CoreFoundation                      -[NSInvocation invokeWithTarget:] + 64",
    "32  CoreFoundation                      ___forwarding___ + 956",
    "33  CoreFoundation                      _CF_forwarding_prep_0 + 96",
    "34  CoreFoundation                      __invoking___ + 148",
    "35  CoreFoundation                      -[NSInvocation invoke] + 424",
    "36  ViewBridge                          __deferNSXPCInvocationOntoMainThread_block_invoke + 132",
    "37  ViewBridge                          __wrapBlockWithVoucher_block_invoke + 56",
    "38  ViewBridge                          deferredBlockOpportunity_block_invoke_2 + 356",
    "39  CoreFoundation                      __CFRUNLOOP_IS_CALLING_OUT_TO_A_BLOCK__ + 28",
    "40  CoreFoundation                      __CFRunLoopDoBlocks + 352",
    "41  CoreFoundation                      __CFRunLoopRun + 2384",
    "42  CoreFoundation                      CFRunLoopRunSpecific + 572",
    "43  HIToolbox                           RunCurrentEventLoopInMode + 324",
    "44  HIToolbox                           ReceiveNextEventCommon + 676",
    "45  HIToolbox                           _BlockUntilNextEventMatchingListInModeWithFilter + 76",
    "46  AppKit                              _DPSNextEvent + 684",
    "47  AppKit                              -[NSApplication(NSEventRouting) _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 688",
    "48  ViewBridge                          __77-[NSViewServiceApplication vbNextEventMatchingMask:untilDate:inMode:dequeue:]_block_invoke + 148",
    "49  ViewBridge                          -[NSViewServiceApplication _withToxicEventMonitorPerform:] + 152",
    "50  ViewBridge                          -[NSViewServiceApplication vbNextEventMatchingMask:untilDate:inMode:dequeue:] + 168",
    "51  ViewBridge                          -[NSViewServiceApplication nextEventMatchingMask:untilDate:inMode:dequeue:] + 100",
    "52  AppKit                              -[NSApplication run] + 480",
    "53  AppKit                              NSApplicationMain + 880",
    "54  libxpc.dylib                        _xpc_objc_main + 816",
    "55  libxpc.dylib                        _xpc_main + 40",
    "56  libxpc.dylib                        _xpc_copy_xpcservice_dictionary + 0",
    "57  ViewBridge                          xpc_connection_handler + 0",
    "58  ViewBridge                          NSViewServiceMain + 360",
    "59  com.apple.appkit.xpc.openAndSavePan main + 156",
    "60  dyld                                start + 6000"
), com.apple.ViewBridge.error.description=NSViewBridgeErrorServiceBootstrap})
```
Based on a quick google I found [this](https://developer.apple.com/forums/thread/782271), which seems to be a similar issue. I could be wrong but perhaps that helps.

m8 V1
macOS 15.4.1
tytools 0.9.9 (downloaded today)

Peterbing, 2025-05-02:
Based on a suggestion from the m8 Discord, deleted and reinstalled the latest and used Terminal to remove the quarantine bit instead of System Prefs. Unsure if this was the issue or if it was just user error the first time. Fixed!

closed 2025-05-02 by Peterbing

felixbrgr, 2025-09-09:
I am struggling with the exact scenario since a while. @Peterbing could you please elaborate on how you achieved the solution?

felixbrgr, 2025-09-09:
I just solved my issue finding the problem was TyUploader not sitting in my apps directory. From there, the upload panel worked just fine!