Koromix/tytools #117: On Mac, file selection dialogs only open if tyUploader is at root of Applications folder

hkleinespel, 2025-09-16:
I have tried putting the executables in a subfolder in the Applications folder, and that results in the file selection dialogs (in my case, firmware hex files) do not open in that case.

Moving the executables to the Applications folder directly resolves the issue.

tyTools 0.9.9
MBA 13" M1
Mac OS 26.0

maxekman, 2026-02-21:
In the meantime I run:

```
/Applications/tycmd upload ~/Downloads/M8Firmware/M8_V6_5_1G_MODEL02.hex
```

lovettbarron, 2026-09-27:
> In the meantime I run:
> 
> ```
> /Applications/tycmd upload ~/Downloads/M8Firmware/M8_V6_5_1G_MODEL02.hex
> ```

Just want to confirm this works on Tahoe 26.6.2, where the UI issue still persists. :) Also for a Dirtywave m8